DLaxV-Community-Projects-Hub/Open-Shot-Clock #31: Controller: Lipos might get damaged when not treated properly

ChangeD20, 2023-04-04:
as a user I want to be sure the shot clock I build is totally safe. When the build in lipos of the controller are not treated and used the right way there is a risk of self-ignition. Also there is a risk the relatively cheap lipos are not build properly.

-> include safety warning in instructions and recommend using only power banks instead

ChangeD20, 2023-04-25:
some controllers in current design state (and previous versions) work without lipo, some are not. Up to know it´s not clear why. The controllers are restarting frequently (about every second).

ChangeD20, 2023-04-25:
safety warning added instructions adapted in the release 1.1.0

cstatz, 2023-04-25:
> some controllers in current design state (and previous versions) work without lipo, some are not. Up to know it´s not clear why. The controllers are restarting frequently (about every second).

This issue might be related to increased current draw from the wifi/rf module.